maxbbraun/pisight #1: Medum paywall

gingerbeardman, 2020-08-28:
Nice work in the project!

It's a shame the article pisight.camera points to is behind the Medium paywall. I had to use a Safari Private Window to read it.

maxbbraun, 2020-08-28:
The domain [pisight.camera](https://pisight.camera) redirects to a link that deliberately has no paywall. It seems to work for most people, but I've had another person mention issues too. Maybe a bug on the Medium side.

closed 2020-08-28 by gingerbeardman

moorlag, 2023-12-28:
[image]
Sadly it's no longer accessible. I've bought all the components, nice Christmas holiday project! Thanks @maxbbraun!

maxbbraun, 2023-12-29:
Try [this link](https://debugger.medium.com/todays-webcams-are-boring-so-i-brought-back-a-classic-291cc7c94c76?sk=20bc3e039f6b39cf1df6cc4f825c8e64) instead. I've also updated the [pisight.camera](https://pisight.camera) redirect, but there might be a caching issue.