maxbbraun/pisight #3: MacBook Pro not detecting Raspberry Pi Zero W

Wthebinder, 2020-09-07:
Thank you so much for making this, I have always wanted to resurrect my iSight Camera!

I have started to order the parts for this project and I am at the point where I can start installing the software. I have both a Raspberry Pi Zero W and a Raspberry Pi Camera Module v2 (and a micro-sd card too). I also have the Anker Micro-USB cord too. 

I started the software part of this project by installing Raspbian and followed the installation instructions via SSH over a Micro-USB Ethernet Adapter. I also connected it to my WiFi network too. After I finished all of the installation steps, I plugged it into my 2017 MacBook Pro 15" running MacOS Catalina (10.15.6). My MBP didn't detect it, even in a "System Report".

I have tried troubleshooting by doing the following:
- Using the Anker Micro-USB cord with an Android Phone and my MBP. It worked.
- Taking pictures with my Raspberry Pi Zero W over SSH.
- Trying different Raspberry Pi Zero Ws.
- Reinstalling the software on my Pis.
- Different Micro-SD Cards

I am not sure how to proceed as I couldn't find anything online. This is my first Raspberry Pi Project so I am sorry if there is something obvious that I am missing.

maxbbraun, 2020-09-07:
What do you see when you run `system_profiler SPUSBDataType`? You might want to give it a few seconds to boot first. It should show something like this:

```
$ system_profiler SPUSBDataType
USB:

    USB 3.1 Bus:

      Host Controller Driver: AppleUSBXHCITR
      PCI Device ID: 0x15ec 
      PCI Revision ID: 0x0006 
      PCI Vendor ID: 0x8086 
      Bus Number: 0x00 

    USB 3.1 Bus:

      Host Controller Driver: AppleIntelCNLUSBXHCI
      PCI Device ID: 0xa36d 
      PCI Revision ID: 0x0010 
      PCI Vendor ID: 0x8086 

        FaceTime HD Camera (Built-in):

          Product ID: 0x8511
          Vendor ID: 0x05ac (Apple Inc.)
          Version: 72.52
          Serial Number: CCG909704RWH34FFE
          Speed: Up to 480 Mb/s
          Manufacturer: Apple Inc.
          Location ID: 0x14600000 / 46
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0
          Built-In: Yes

        PiSight:

          Product ID: 0xdead
          Vendor ID: 0x05ac (Apple Inc.)
          Version: 0.01
          Serial Number: 100000000d2386db
          Speed: Up to 480 Mb/s
          Manufacturer: Apple
          Location ID: 0x14300000 / 55
          Current Available (mA): 500
          Extra Operating Current (mA): 0

```

I usually test with QuickTime, which seems to be the most resilient video preview (rather than Photo Booth, for example, which sometimes doesn't work with the camera). Go to File > New Movie Recording and then select PiSight from the dropdown next to the red record button. That should show a video preview from the PiSight.

If none or only one of these works, it might be a clue.

Wthebinder, 2020-09-08:
Thanks for your reply! Unfortunately, the raspberry pi doesn't show up when I do _system_profiler SPUSBDataType_.

maxbbraun, 2020-09-09:
Did you try the other USB ports? Which USB version shows up for the bus?

Wthebinder, 2020-09-09:
Thanks for your reply!

I have tried all 4 of my USB-C Ports using a handful of known good USB-C to USB-A adapters, to no luck. (I waited until I can SSH into it over WiFi, which tells me when it is done booting.) I used the same micro-USB cord earlier today to get some pictures off of an android device. My MacBook Pro has four USB 3.1 Buses.

Could it because I am using a different version of macOS?

maxbbraun, 2020-09-13:
Not sure. Maybe [this](https://medium.com/@kabutosan/so-after-two-days-of-troubleshooting-i-have-identified-to-main-issues-ae46ef05350) is helpful?

Wthebinder, 2020-09-15:
Hmmm, maybe it is my micro-usb cable? Is the one listed on here known to work for this project?

maxbbraun, 2020-09-15:
Yes, the one listed works for me.