maxbbraun/pisight #4: Assembly difficulties

John-K, 2020-09-27:
Can you share any photos or experiences from your assembly?

Assembly issues:
1. I can't figure out how to install the camera module in the top of the print, even without the flex attached. It seems like the camera sensor housing is meant to slip through the gap in the ring, but the screw bosses interfere with components on the bottom of the camera board.
2. Does your "Apple medallion" fit in place at the back? The camera flex appears to block it as it needs an extra bit of room to do a 90 degree bend.

Things that work well:
1. The RPiZero fits in nicely and the screws hold it neatly in place.
2. The Front lens assembly fits snugly into the camera ring and the iris actuates nicely.

maxbbraun, 2020-09-27:
Glad to hear you're trying it out!

> I can't figure out how to install the camera module in the top of the print, even without the flex attached. It seems like the camera sensor housing is meant to slip through the gap in the ring, but the screw bosses interfere with components on the bottom of the camera board.

Getting the camera in is definitely the trickiest part. I ended up having to bend the front ring a bit and push the camera through. Definitely test the camera after that to make sure it hasn't been damaged. This is obviously not a great solution, so I've been playing with a version of the model that makes the front ring just a little narrower to give the camera enough room to be inserted, but that isn't ready yet. You could try filing off some material to achieve the same effect.

> Does your "Apple medallion" fit in place at the back? The camera flex appears to block it as it needs an extra bit of room to do a 90 degree bend.

That had worked for me, although I did have to bend the camera flex very tightly.

John-K, 2020-09-28:
I was able to finally get the camera module installed by snipping off the top-left corner of the camera PCB where the mounting hole is in the PCB and then by pushing a fair bit on the ring.

cooperpellaton, 2020-10-16:
I tried installing the camera today. 

Clearance is _really_ snug. Managed to break one of the SMDs near the cable mounting point which subsequently broke the camera. I ordered the same print material/resolution from Fictiv as @maxbbraun and found that there is a decent amount of play in the upper ring. In my second attempt, I'll try to flex the edge rather than come at it from the side (90°) approach.

On the positive side, the fit of the rest of the components is perfect. Was a bit difficult to get the ribbon cable into the recess but now that it's in there it's not going anywhere.
 
[image]

Lastly, a word of warning - the screws on the _inside_ assembly: the frame attachment of the black plastic, are _really_ tight. It stripped the bit I was using!

moorlag, 2024-02-13:
Can you share a routing overview of the flex cable?

maxbbraun, 2024-02-13:
It's been a while since I last did this, but maybe these pictures help. The flex cable is folded diagonally behind the camera.

[image]

[image]

moorlag, 2024-02-13:
wow thanks @Max Braun!
Happy that I have a few cables to get it right.

It's such a fun project :-) thanks again!

maxbbraun, 2024-02-13:
I'm glad you're enjoying it! :)