maxbbraun/pisight #5: Field of view issues

samnunn, 2020-09-27:
Hi Max!

Awesome project mate. I've bought a second-hand iSight and I'm in the throes of recreating this setup for myself. One question though: why did you switch down to 720p?

I used Dave's original 1080 setup and I noticed that the field of view was quiet a lot narrower than expected. Is this related? I haven't actually gotten your fork working yet, so I'm yet to compare the pair.

Many thanks.

Sam

maxbbraun, 2020-09-27:
Hi Sam!

Yeah, I had some issues with the uvc-gadget implementation and I think field of view was part of it. Overall, it felt like the code was written for 720p anyway (e.g. [this](https://github.com/climberhunt/uvc-gadget/blob/3d1e49e420fd2359cf95e170336144189eaaad06/uvc-gadget.c#L2047)) so I reverted it to that and never got back to trying to get 1080p or more to work. I'm not sure if anything would be fundamentally in the way of that though.

Let me know how it goes!
Max

John-K, 2020-09-28:
Per https://www.raspberrypi.org/documentation/raspbian/applications/camera.md the Camera Module
[image]
It would appear that 1640x922 would be the sweet-spot with full FOV and 2x2 pixel binning.

samnunn, 2020-10-02:
Awesome notes folks! I'm a little lost about how I would implement 1640x922 successfully, though. Any pointers?

iliesaya, 2020-10-12:
I agree that it would be nice to have the full field of view of the camera

muffinresearch, 2021-01-08:
I had similar issues with zoom randomly switching to a higher resolution (which showed as very zoomed in) during conf calls using [showmewebcam](https://github.com/showmewebcam/showmewebcam) v1.70. [Following some helpful directions from @htruong who mentioned a way of enabling github actions on the fork to enable image builds there, I patched the list of available resolutions to fix it to 1280x720 here](https://github.com/muffinresearch/showmewebcam/pull/1). The built images from that PR are available by clicking details on the relevant build and looking at the artifacts dropdown. Hopefully at some point more control for resolutions will be exposed to users without the need to create a new image restricting resolutions.

FWIW: I found getting the sd card out to re-flash it was possible without removing the camera module by unscrewing the zero and carefully hinging the sd card end to be able to get to it. At some point showmewebcam will probably facilitate updates via usb. See https://github.com/showmewebcam/showmewebcam/issues/54

rickyktan, 2021-08-15:
> I had similar issues with zoom randomly switching to a higher resolution (which showed as very zoomed in) during conf calls using [showmewebcam](https://github.com/showmewebcam/showmewebcam) v1.70. [Following some helpful directions from @htruong who mentioned a way of enabling github actions on the fork to enable image builds there, I patched the list of available resolutions to fix it to 1280x720 here](https://github.com/muffinresearch/showmewebcam/pull/1). The built images from that PR are available by clicking details on the relevant build and looking at the artifacts dropdown. Hopefully at some point more control for resolutions will be exposed to users without the need to create a new image restricting resolutions.
> 
> FWIW: I found getting the sd card out to re-flash it was possible without removing the camera module by unscrewing the zero and carefully hinging the sd card end to be able to get to it. At some point showmewebcam will probably facilitate updates via usb. See [showmewebcam/showmewebcam#54](https://github.com/showmewebcam/showmewebcam/issues/54)

@muffinresearch I have tried to rebuild the showmewebcam using the same resolution restrictions as you (1280x720), but have run into multiple bumps. Are you able to share the built images again? The artifacts in your pull request have expired, and one can't download the images anymore.