maxbbraun/pisight #8: Camera only works when MacBook Pro lid is opened

weslebsack, 2020-10-18:
I built a PiSight that works great when the lid of my MacBook Pro is opened. However, as soon as I close the lid, the camera shuts off – just like the internal FaceTime camera. I went into the System Report, and it appears the PiSight and FaceTime camera have the same VendorID (1452). Because of that, I'm wondering if macOS thinks the PiSight is a FaceTime camera, and that's why it behaves like one?

Is it possible to change the VendorID of the camera, so that macOS doesn't (apparently) confuse it with being a FaceTime camera?

maxbbraun, 2020-11-05:
You could change the Vendor ID [here](https://github.com/maxbbraun/uvc-gadget/blob/7b504dec9a95504b6880033936165a7eb3684d78/multi-gadget.sh#L4), but I doubt that's the reason for this behavior. Does the same happen with any USB port?

weslebsack, 2020-11-05:
I also noticed that Apple is listed as the manufacturer here, which could be contributing to the confusion by the OS:

https://github.com/maxbbraun/uvc-gadget/blob/7b504dec9a95504b6880033936165a7eb3684d78/multi-gadget.sh#L15

It happens on every USB port. The camera works when the MacBook Pro’s lid is open, and as soon as I close it, the camera turns off. This behavior is the same as when people around the internet have taken an actual FaceTime camera from an old MacBook Pro or iMac and spliced into its USB pins to make it a USB cam. The OS also thinks it’s an internal camera and shuts it off when the lid is closed.

maxbbraun, 2020-11-06:
You should be able to test that theory by changing those properties to something else.

I also just tried what happens when I close the lid on my MacBook (16" 2019, Catalina) and I can't reproduce the issue you're seeing. The camera stays up.

conlank, 2020-11-17:
I had this same problem. I edited the multi-gadget.sh to replace 
`echo 0x05ac > /sys/kernel/config/usb_gadget/pi4/idVendor
echo 0xdead > /sys/kernel/config/usb_gadget/pi4/idProduct
echo 0x0001 > /sys/kernel/config/usb_gadget/pi4/bcdDevice
echo 0x0200 > /sys/kernel/config/usb_gadget/pi4/bcdUSB`
with the segment below.
`echo 0x1d6b > /sys/kernel/config/usb_gadget/pi4/idVendor
echo 0x0104 > /sys/kernel/config/usb_gadget/pi4/idProduct
echo 0x0100 > /sys/kernel/config/usb_gadget/pi4/bcdDevice
echo 0x0200 > /sys/kernel/config/usb_gadget/pi4/bcdUSB`

After restarting the PiSight, I was able to have the camera continue to function while in clamshell mode on my MacBook Air.

snake302, 2022-02-21:
> I had this same problem. I edited the multi-gadget.sh to replace `echo 0x05ac > /sys/kernel/config/usb_gadget/pi4/idVendor echo 0xdead > /sys/kernel/config/usb_gadget/pi4/idProduct echo 0x0001 > /sys/kernel/config/usb_gadget/pi4/bcdDevice echo 0x0200 > /sys/kernel/config/usb_gadget/pi4/bcdUSB` with the segment below. `echo 0x1d6b > /sys/kernel/config/usb_gadget/pi4/idVendor echo 0x0104 > /sys/kernel/config/usb_gadget/pi4/idProduct echo 0x0100 > /sys/kernel/config/usb_gadget/pi4/bcdDevice echo 0x0200 > /sys/kernel/config/usb_gadget/pi4/bcdUSB`
> 
> After restarting the PiSight, I was able to have the camera continue to function while in clamshell mode on my MacBook Air. I did not test these changes one at a time, so I can't pinpoint which change exactly solved the issue.

Oh man, you saved my day, was trying to figure what's going on wrong.