maxbbraun/pisight #9: Gray screen on MacBook 2010

lewixlabs, 2020-10-22:
Hi!
First congrats for your great project 👍 

I built from scratch step by step PiSight project:

1. flashed pi os image to my pi zero
2. enabled camera & serial interfaces from raspi-config
3. cloned your project and launched your bash script

Results:
- on my old MacBook 2010 (High Sierra) cam stream show a "gray" only screen in whereby or other videocall systems (Ms Teams crashes)
- trying with Parallels in a debian based virtual machine camera streams perfectly works.

I tried to change idVendor from Apple to another (for example TP-Link, listed here https://usb-ids.gowdy.us/read/UD/) but with same result: on my Mac PiSight doesn't work.

Could you suggest me what i could check? Maybe some logs?

Anyway thanks again!
Lewix

maxbbraun, 2020-11-05:
Thank you! Glad you're trying it out.

Whether you have issues with the camera stream may also depend on the piece of software you're running to consume it. I've found QuickTime to be most reliable. Maybe try that?

lewixlabs, 2020-11-27:
@maxbbraun today I noticed an update of Dave Hunt project:
https://github.com/showmewebcam/showmewebcam
https://www.youtube.com/watch?v=nH2G16YoBT4&feature=push-fr&attr_tag=ZatzxTeeD8FzI4M5%3A6
This one has also a ready to go Pi0(W) image (little, it fits on a 1GB sd card) and it works on my old MacBook 2010!

I keep open this issue for info but feel free to close it.

Hi!
Lewix

maxbbraun, 2020-12-28:
Glad that worked for you! I added a reference to `showmewebcam` in 1c28d13334b7542b2ae6a69a3bf8b8be25226221.

closed 2020-12-28 by maxbbraun