maxbbraun/pisight #10: Post-install image

j-james, 2020-10-27:
Hey maxbbraun,

In what's probably a mistake in hindsight, I decided to save $5 on the Zero by not purchasing the Wifi version, forgetting that I lack the adapters to connect a monitor and keyboard to the Pi. This meant I had to jump through some hoops to get the Pi set up which don't seem to have worked...

By any chance, would you be able to supply an image of the operating image post-install? I've been unable to get the Pi to show up as a USB device - probably related to me running the installation script in a `chroot`.

Thanks a bunch!
j-james

maxbbraun, 2020-11-16:
Try [this](https://drive.google.com/file/d/16xw0Vxwv2uJkPhz652Kn_8f1cDQ6GItg/view?usp=sharing)!

j-james, 2020-11-16:
Hey, thanks! 

(I did manage to fix my problems with the `chroot` and get it set up that way, but I'm sure someone else will run into this)

closed 2020-11-16 by j-james