maxbbraun/pisight #11: Completed Build

cooperpellaton, 2020-11-03:
Not sure where to put feedback/comments. Maybe a feedback section in the wiki?

# Overview
On balance, this went well. I'd recommend purchasing at least 2x the number of ribbon cables you anticipate since they're extremely easy to puncture when working the excess into the case-area recess. Further, I think V2 of the model could do away with the strut supporting the upper ring on the (looking directly-on) left-hand side. There's enough flex in the print that without it you can easily slip the camera underneath. In terms of camera-module sizing, I found nail clippers work extremely well. I found a pair with a curved blade that allowed me to match the corner of the camera module PCB to the case housing. 

Total time (including a re-order of camera module + ribbon cables after breakage) was 4 weeks from inception to completion. 

# Glamour Shots
[image]
[image]

maxbbraun, 2020-11-05:
Very cool. Thank you for sharing, Cooper!

tillycatcat, 2020-12-20:
Hey @cooperpellaton - I'm just about to send the print to the chap with a 3d printer to print. You mentioned the strut (and I'm a complete beginner here). Is it possible to send a picture/a model file of what you mean?! I'll definitely give it a go! 

Thanks
Geoff.

cooperpellaton, 2020-12-22:
@tillycatcat if you look in the 3D print where the camera module should go there are 2 struts that support the upper ring. If, looking head one, you remove the _left_ strut, the structural integrity should be fine **and** you should be able to get the camera inside more easily.

grantland, 2021-02-23:
I'm in the process of building one myself and have noticed that the ribbon cable pokes out the back where the original white cap with the Apple logo is supposed to go. Is it safe to bend the cable to get the cap on or did you leave it without the cap on?

maxbbraun, 2021-02-28:
@grantland I had to bend the cable very close to the connector, which isn't ideal, but doesn't seem to have caused any issues so far.

cooperpellaton, 2021-02-28:
@grantland I used [this](https://www.waveshare.com/raspberry-pi-zero-v1.3-camera-cable.htm) cable and had no problems with it being too long. It did, however, have to bent in order for it to fit into the printed recess.

tillycatcat, 2021-03-02:
I've gone through a few cables now trying to figure out the exact point where to bend it. I wonder if the cable will withstand a fold?

cooperpellaton, 2021-03-03:
I punctured one cable when bending but that was due to applying too much force with a pair of tweezers. I think as long as you don't overly force it then there won't be any problems.

tillycatcat, 2021-03-14:
I'm going to have another crack at this soon. Has anyone got a picture of the bend at the camera module?

austriker27, 2021-11-11:
Looks stunning! @cooperpellaton I'm curious, I'm curious about buying a completed build - would you be interested in building one and selling it?

cooperpellaton, 2021-11-16:
@austriker27 I'd be open to doing it, however, I'd caution you that the cost will likely be prohibitively expensive. Do feel free to email me.

austriker27, 2021-11-17:
@cooperpellaton yeah i figure it won't be cheap but I'm still curious. What's your email?

cooperpellaton, 2021-11-17:
@austriker27 see my website linked in my bio.