maxbbraun/pisight #12: FDM Printing?

mattkerrison, 2020-11-15:
I've bought a used iSight & have the Raspberry pi lying around, but I was curious if anyone has tried to print the enclosure with an FDM printer? 
With a 0.2mm nozzle I'd think it would likely be possible, curious as I would prefer that than ordering a resin printed piece.

phenomeus, 2020-12-15:
same for me and also I bought an ender 3 v2 and will try to print this. right now I play around with Cura slicer. will report asap, but right now I have too much projects going on in parallel

stockHuman, 2021-01-26:
_Just_ finished completing this project, printed this on an Ender 3.

I had to omit the tiny fasteners due to the very tight tolerances needed around the shutter assembly & the two anchoring points to the cylindrical case. It's effectively friction-fit, but that works rather well. A dremel comes in handy.
[image]

phenomeus, 2021-01-26:
what settings on the ender? standing upright?

what adhesion setup? skirt? brim? raft? 

infill? suport?

so many questions :D I will do it on my ender 3 v2 asap, but I still have so many projects going on

stockHuman, 2021-01-28:
@phenomeus 

That's right, standing tall, printed with the camera-insert side touching the build plate. I did this to minimize supports, which were a pain to remove.

Simple skirt around the print, no special adhesion settings. And I believe 14% infill (gyro), nearly default (Cura HQ) settings otherwise. No part cooling fan, but that's specific to my machine.

Good luck on your print. The real pain was assembly!

phenomeus, 2021-03-31:
finally I got it printed on my ender 3 v2 with 10% support that I had to pull off. see pictures, everything went fine
but I had the camera side pointing to the top. 

[image]
[image]
[image]
[image]
[image]
[image]

msgilligan, 2021-03-31:
> with 10% support that I had to pull off. see pictures

Do you have any photos with the support still attached? I'd be curious to see what it looked like at that point.

My Ender 3 V2 is supposed to arrive today and I am new to 3D printing, but this is one of the first projects I'm going to try.

phenomeus, 2021-03-31:
sorry I don't but here you can see the green "helpers" = support that I had to remove 

[image]
[image]

msgilligan, 2021-03-31:
@phenomeus Thanks!

phenomeus, 2021-03-31:
your welcome, here you can see my finished build https://github.com/maxbbraun/pisight/issues/16