maxbbraun/pisight #13: Activity LED using the original iSight LED

muffinresearch, 2021-01-09:
Thanks for the excellent project. After getting a build working I've been playing around with adding an LED to show when the camera is active.

[Showmewebcam handily switches GPIO 21 to `HIGH` when the camera is in-use](https://github.com/showmewebcam/showmewebcam#led-indicator) so it's possible to put an LED in the place of the one that would have been on the original iSight.

All that's required is a 330ohm resistor and an LED. As seen in this test run:

[image]

I pulled out the original LED from the iSight and carefully soldered two cables either side of the surface mount LED. I used some shrinkwrap to shield the connections from the case and used shrinkwrap with a hole punched in it to aim to get the light only going up into the small clear plastic diffuser. A small amount of rearwards leakage was fixed with some Aluminum tape on the body.

Here's the end result:

[image]

maxbbraun, 2021-01-10:
Very cool!

grantland, 2021-02-23:
@muffinresearch Super cool! Does the original LED need a 330ohm resistor too or was it just plug and play?

muffinresearch, 2021-02-23:
> @muffinresearch Super cool! Does the original LED need a 330ohm resistor too or was it just plug and play?

Yep, I added the same size resistor with the original led too.