maxbbraun/pisight #17: New Dell webcam might be a perfect fit for PiSight

realph, 2021-06-29:
Dell just announced a [new 4K webcam](https://www.tomshardware.com/uk/reviews/dell-ultrasharp-webcam) that is heavily inspired by the original iSight design. The good news is we may be able to fit this camera module inside of an iSight body. The bad news is the casing on the Dell webcam seems to be bigger than that of the iSight...

[image]

I've never really done any hardware hacks before. Is this a simple case of pulling out the Dell camera module and seeing it if fits inside the iSight tube? If It does I assume we'll need to also need to make sure there's enough room for a 3D printed frame like Max's original design so it fits snug.

Does anyone have any additional advice on how to go about this? I may try and pick one of these webcams up over the weekend and tinker around with it to verify if it fits.

realph, 2021-06-29:
Hmm it looks like the board touches the edges of the anodised frame so it can't be much smaller than 420mm, which means it won't fit inside the iSight body. 😢 

[image]

mattkerrison, 2021-06-29:
Nice idea but I think @realph is right

cwelsh1812, 2021-06-30:
Think you have an extra '0' on the measurements, otherwise that is one BIG webcam !

Would be great if it fitted but the search continues I think.