maxbbraun/pisight #19: Script doesn't work with Raspberry pi zero 2 w

Emstr, 2022-04-02:
Hey Max. Big fan.

I'm working on this project and can only get my hands on a pi zero 2 w, rather than the original. Unfortunately when I use your script, the camera feed shows up for half a second, and then the camera becomes unresponsive. Unfortunately showmewebcam hasn't yet added support for zero 2 w. I wonder if you have any ideas?

maxbbraun, 2022-04-02:
Hm, I don't have a Pi Zero 2 W to check. Did you ask over at [showmewebcam](https://github.com/showmewebcam/showmewebcam/issues) yet?

Emstr, 2022-04-03:
Yep – looks like they are working on support but it's not ready yet :(

Yours seems like it works better for me, but then seems to crash. Is there any kind of gpu/memory setting I could try to tweak?

maxbbraun, 2022-04-04:
If they're already working on it, that's probably your best bet!

moorlag, 2024-02-13:
I've found a solution using the FULL install and not the remote only. 
Some how it does work with a full installation of the OS.

olm3ca, 2024-11-16:
> I've found a solution using the FULL install and not the remote only. Some how it does work with a full installation of the OS.

Could you clarify which full install you used, @moorlag ? 64 or 32 bit, Legacy or regular? Thanks

vguttmann, 2025-01-25:
I've made a working script and PR to this repo, in case anyone is still interested

maxbbraun, 2025-01-26:
@vguttmann's PR is https://github.com/maxbbraun/pisight/pull/25, for anyone who wants to follow along.