maxbbraun/pisight #22: Camera Module v3 support

acketon, 2023-01-11:
The v3 Camera Module that was recently released looks like a potential upgrade opportunity. Has anyone attempted to use it yet?

acketon, 2023-01-11:
https://www.raspberrypi.com/products/camera-module-3/

applecuckoo, 2023-04-01:
@acketon @maxbbraun the people at Raspberry Pi upgraded the project themselves, check it out [here](https://www.raspberrypi.com/news/how-to-build-your-own-raspberry-pi-webcam/?_wem_rev_track=61241559-855e-41e4-937a-1527a21c8ce3%3A89f06ed5-f7d0-49e1-be81-20fa2f319a6b)

acketon, 2023-07-11:
I tried the tutorial above but the performance on a Raspberry PI Zero W (1st gen) is choppy and unusable video. I haven't been able to get a Pi Zero 2 due to shortages. I'm not sure if there is a way to improve the performance... its very different than my other PiSight setup using the Pi Zero 1 w/Camera Module 2. But they are also using different software and setup approaches so I'm not sure if that is part of the issue.

It looks like the ShowMeWebCam project doesn't yet have support for Camera Module 3 yet.