maxbbraun/pisight #26: How can I have the PiSight show up as "PiSight" on host device?

elcalzado, 2025-03-08:
It shows up as "UVC Camera" on my Windows 11 PC and I want to change what the name of it is. I can't figure out how to. The only place I found "UVC Camera" mentioned was in /sys/kernel/config/usb_gadget/g1/functions/function_name and some other directory within usb_gadget/ but those files seem to not be editable. I've gone through the uvc_gadget's repo and can't find a mention of "UVC Camera". Is my PC just choosing to name it that?

I changed this from the script on the tutorial hoping it would change the name but it did not.
CONFIGFS="/sys/kernel/config"
GADGET="$CONFIGFS/usb_gadget"
VID="0x05ac"
PID="0xdead"
SERIAL="0123456789"
MANUF="Apple Pi"
PRODUCT="PiSight"
BOARD=$(strings /proc/device-tree/model)
UDC=\ls /sys/class/udc` # will identify the 'first' UDC`

maxbbraun, 2025-03-09:
It's odd, because the custom name is being set by the script. I can't test it out on Windows, unfortunately.

elcalzado, 2025-03-09:
That's what I thought.. I can't test it on Mac until tomorrow. From what you know though, is the PRODUCT variable all that needs to be changed to make it show up as something? If so, is it most likely an issue with how Windows is detecting it? I'll leave a [picture](https://imgur.com/a/bQ3MGUq) of the info I see on Device Manager for the camera.

elcalzado, 2025-03-21:
In case anyone is still struggling with this I just figured it out. Not really sure why this works like that but I'll share my findings.

If you add `echo "PiSight" -n  | sudo tee functions/$FUNCTION/function_name` to the script after `mkdir functions/$FUNCTION` it should work!

Seems like by default the linux driver creates a file named function_name with value "UVC Camera". Before, I didn't know how to change this since I would get a "device is using this" or similar error. It seems to be that was because I was trying to do it after the symlink between functions/ and configs/ but I didn't test to make sure.

closed 2025-03-21 by elcalzado

maxbbraun, 2025-03-23:
Hi @elcalzado! Could you check if you can run the full setup with a patched in PR #27 on a fresh Pi Zero 2 W without any errors? Then we can merge the fix.

elcalzado, 2025-03-24:
Sure! I’ll get it done some day this week but I’ll be away from my PC for a the first few days of the week. I’ll notify when I do it.